moveit/moveit_msgs #18: Indigo release 0.7.1

130s, 2016-04-13:
This seems to be blocking https://github.com/ros-planning/moveit_ros/issues/662

130s, 2016-04-13:
Done https://github.com/ros/rosdistro/pull/11078

closed 2016-04-13 by 130s

130s, 2016-04-14:
Almost 12 hours after https://github.com/ros/rosdistro/pull/11078 was merged, [debs are not yet soaked to shadow repo](http://repositories.ros.org/status_page/ros_indigo_default.html?q=moveit_msgs). @tfoote Is this normal to take this long?

tfoote, 2016-04-14:
For something like moveit_msgs yes it's normal. It has 20+ direct dependencies and the dependencies can be 6-7 layers on top of moveit_msgs with some of those packages taking close to an hour to build (some approach 10 hours to build on arm platforms). Across all architectures you can see that the buildfarm has been churning continuously with 12 executors since that release. http://build.ros.org:8080/label/buildslave/load-statistics

The more dependencies of any package released the more time it takes for things to rebuild. That's why we batch up low level releases for simultaneous rebuilds. 

I'd expect the first architectures to come out tonight.

130s, 2016-04-14:
I see, thanks for the detail!
Now I see moveit_msgs are available on shadow-fixed.