moveit/moveit_msgs #22: Release moveit_msgs

davetcoleman, 2016-06-14:
@mikeferguson @IanTheEngineer @130s Can we do a release of moveit_msgs in indigo for [this](https://github.com/ros-planning/moveit_ros/pull/686) new feature?

130s, 2016-06-14:
Ok, I'm running prerelease tests for I & J locally with 2 level downstream. I'm sure it'll take time.

v4hn, 2016-06-14:
130s: good point mentioning jade, I just cherry-picked the changes to moveit_msgs and moveit_core to their respective jade-devel branches. (I suppose it's valid to cherry-pick commits that have been reviewed to other branches unless they might break something there?)

Probably too late for your tests though @130s

130s, 2016-06-14:
@v4hn no worries, I just re-triggerred the script (2 commands). 

Note that now I have a slight suspicion how effective ROS Prerelease test given the issue https://github.com/ros-infrastructure/prerelease_website/issues/31. We may or may not need to pay too much attention to the result of my Prerelease Test for now.

davetcoleman, 2016-06-14:
@v4hn feel free to cherry pick to *newer* branches without review, yes. but lets PR backports

130s, 2016-06-15:
Indigo passed. Jade prerelease test fails at downstream (log pasted below). I am +1 for making a release since this failing one is a leaf.

```
<== Finished processing package [10 of 27]: 'jsk_pcl_ros_utils'

==> Processing catkin package: 'jsk_pcl_ros'
==> Building with env: '/tmp/catkin_workspace_overlay/devel_isolated/jsk_pcl_ros_utils/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/tmp/catkin_workspace_overlay/build_isolated/jsk_pcl_ros'
==> make -j1 tests in '/tmp/catkin_workspace_overlay/build_isolated/jsk_pcl_ros'
Scanning dependencies of target gtest
Building CXX object gtest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
Linking CXX shared library libgtest.so
Built target gtest
Scanning dependencies of target test_extract_indices
Building CXX object CMakeFiles/test_extract_indices.dir/test/test_extract_indices.cpp.o
Linking CXX executable /tmp/catkin_workspace_overlay/devel_isolated/jsk_pcl_ros/lib/jsk_pcl_ros/test_extract_indices
Built target test_extract_indices
Scanning dependencies of target install_test_data
[jsk_pcl_ros] Start installing test data
Traceback (most recent call last):
  File "/tmp/catkin_workspace_overlay/src/jsk_pcl_ros/scripts/install_test_data.py", line 77, in <module>
    main()
  File "/tmp/catkin_workspace_overlay/src/jsk_pcl_ros/scripts/install_test_data.py", line 70, in main
    compressed_bag='2015-11-04-19-37-29_baxter-kiva-object-in-hand-cloud/vision.compressed.bag',
  File "/tmp/catkin_workspace_overlay/src/jsk_pcl_ros/scripts/install_test_data.py", line 43, in install_test_data
    os.mkdir(ros_data_dir)
OSError: [Errno 2] No such file or directory: '/home/buildfarm/.ros/test_data'
make[3]: *** [CMakeFiles/install_test_data] Error 1
make[2]: *** [CMakeFiles/install_test_data.dir/all] Error 2
make[1]: *** [CMakeFiles/tests.dir/rule] Error 2
make: *** [tests] Error 2
<== Failed to process package 'jsk_pcl_ros': 
  Command '['/tmp/catkin_workspace_overlay/devel_isolated/jsk_pcl_ros_utils/env.sh', 'make', '-j1', 'tests']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /tmp/catkin_workspace_overlay/build_isolated/jsk_pcl_ros && /tmp/catkin_workspace_overlay/devel_isolated/jsk_pcl_ros_utils/env.sh make -j1 tests

Command failed, exiting.
# END SUBSECTION
```

130s, 2016-06-15:
Ok, I'm on this for I&J.

130s, 2016-06-15:
I https://github.com/ros/rosdistro/pull/11846
J https://github.com/ros/rosdistro/pull/11845

closed 2016-06-15 by 130s