moveit/moveit_msgs #38: Enable pull request testing by ROS buildfarm

130s, 2017-04-14:
Can **any admin of this repo** help to enable the pull request testing by only a few clicks? Please follow [this instruction](http://wiki.ros.org/buildfarm/Pull%20request%20testing#Link_Jenkins_and_your_.21GitHub_repository) ("Recommended" section).

Other chores is already taken care of (e.g. in https://github.com/ros/rosdistro/pull/14529).
Once it's done, then pull request testing powered-by ROS buildfarm will become available.

130s, 2017-04-15:
@dirk-thomas or any admins, could you take care of this? Thank you in advance.

dirk-thomas, 2017-04-15:
Please see https://github.com/ros/rosdistro/pull/14530#discussion_r111661080 I already added the ROS buildfarm user to the `ros-planning` org unit.

closed 2017-04-15 by dirk-thomas