moveit/moveit_msgs #42: Target offset field in position constraint msg

ptriantd-ocado, 2018-01-03:
I was wondering what is the purpose of this field. The Position Constraint msg defines a region and a link of the robot that should remain inside this region. So, there is no target position involved, is there?

JeroenDM, 2020-06-11:
This questions has been here a while, but for future generations:

The position constraint will keep the center of the specified link frame inside the constraint region. Optionally, you can specify an offset. Then it is not the center of the link frame, but the offset point, expressed relative to the link frame, that has to stay inside the constraint region.

It is implemented [here](https://github.com/ros-planning/moveit/blob/272eaa01064df0b0184e2ad8766751ea27ef6c92/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp#L441).

This seems usefull to me when you want to constrain the tip of an end-effector. If the link frame of the end-effector is not centered at it's tip, and there is no other link defined in the URDF at the tip of the end-effector, then you can use this offset.

closed 2020-06-11 by ptriantd-ocado

v4hn, 2020-06-11:
> This seems useful to me when you want to constrain the tip of an end-effector.

More specifically it can account for constraints on objects attached to the eef without requiring the information about the attached object anymore. That is the prime use of the field internally.