moveit/moveit_msgs #71: Melodic Release

tylerjw, 2020-05-06:
What is needed from us to get the melodic-devel branch of this released in the next sync?

rhaschke, 2020-05-07:
We just need to prepare a new release... 
Looking at the diffs since the last release, I'm wondering why we have the subframes support in moveit_msgs, but MoveIt's melodic branch isn't supporting that. I think, we should remove the sub_frames support here before releasing into Melodic.

v4hn, 2020-05-07:
> Looking at the diffs since the last release, I'm wondering why we have the subframes support in moveit_msgs,
> but MoveIt's melodic branch isn't supporting that.

That was pointed out a number of times. During backporting I also explicitly stated that I left this feature out,
because it is distributed over quite a number of commits and was still buggy.
Actually, I still think it's buggy until we change the frame semantics for world object frames with Felix's wip change...

> I think, we should remove the sub_frames support here before releasing into Melodic.

If this change was really not released previously (I thought so) and is the only change to the message, I fully agree!

v4hn, 2020-05-07:
Apparently this change was really never released in melodic and I would veto to remove it there.

felixvd, 2020-05-08:
IIRC the subframes feature was merged only into `master` and not `melodic-devel` because moveit_msgs shouldn't change after release. Am I missing something?

v4hn, 2020-05-08:
Am I missing something?
Yes, the fact that [the message change was indeed merged in `melodic-devel`](https://github.com/ros-planning/moveit_msgs/pull/50) a year ago, although the code support was never added to the main moveit repo.

felixvd, 2020-05-09:
Right. I never changed the target of that PR to master. Splitting the code across multiple repositories has its pitfalls :0

I have no problem with reverting it in melodic-devel and keeping the feature in master/noetic as planned.

tylerjw, 2020-05-10:
I updated the list of potential backports to include only those that add new messages and not ones that change existing messages.  Please advise on which ones I should prepare a backport for?

v4hn, 2020-05-11:
- Add service to update pointcloud octomap (#66)

The corresponding support in https://github.com/ros-planning/moveit/pull/1984 is not merged yet, so no need to backport (yet?)

- Add messages to plan for sequences (#65)

@jschleicher do you want to see these messages in melodic-devel too?
I don't think there is harm in having them and even your standalone repository could use them.

- bump cmake version (#67)

This version is supported since kinetic, so this should be backported.

ct2034, 2020-05-11:
Yes, we need the messages in melodic-devel.

tylerjw, 2020-05-12:
@JafarAbdi It is now your turn to do the next steps.  See Making a new release section of https://moveit.ros.org/documentation/contributing/pullrequests/

tylerjw, 2020-05-20:
@JafarAbdi Prepared the release notes and a version bump: https://github.com/ros-planning/moveit_msgs/pull/77

rhaschke, 2020-05-21:
Release PR: ros/rosdistro#24992

tylerjw, 2020-05-24:
@rhaschke your PR to rosdistro got merged.  What are the remaining steps?  I'm not sure I have the right list of steps in the description for this.

rhaschke, 2020-05-25:
We can already use the newly released package in all builds referring to the testing repos, i.e. `ros-shadow-fixed` in Travis configs. If the next sync happened, the package will be available in the default repo as well. This can be checked here:
http://repositories.ros.org/status_page/ros_melodic_default.html?q=moveit_msgs

closed 2020-05-25 by rhaschke