moveit/moveit_msgs #86: Any plan for the noetic release

rjshim, 2020-08-14:
Hi, 

It seems this repository hasn't been released for the noetic version yet.
Any concrete plans for it?

Ryan

tylerjw, 2020-08-19:
Hello Ryan,

I'm planning on working on this today.  I'm going to create a noetic-devel branch and release that unless there are objections.

v4hn, 2020-08-19:
Do you need to split from `master`? For the main repo we did not want to split yet, though [we do have a preliminary branch](https://github.com/ros-planning/moveit/issues/2253) already.

v4hn, 2020-08-19:
It might make sense to merge #58 and fix the corresponding pr in moveit general before releasing the messages, but I will not spend time on this. Nor did anyone else consider it worthwhile to work on it for quite a long time.

tylerjw, 2020-08-19:
Good point.  I didn't understand the reason for the noetic-devel branch on moveit.  I am in favor of just releasing Noetic on master.  I can look into that issue to see if I can get that finished quickly.  I don't remember what is blocking it.

tylerjw, 2020-08-19:
@v4hn [It doesn't look like we think the approach to the issue you linked is the correct solution.](https://github.com/ros-planning/moveit/pull/1710#issuecomment-557538087)

That being said there will always be changes in flight.  I think we should release Noetic and accept that we can release it again in some time with some breaking changes.  

Noetic is the last ROS1 release so we can no longer rely on future ROS1 releases to provide opportunity for breaking changes and will just have to deal with and accept that that will be a part of the releases unless we just choose not to release any more.  We should be able to use version changes to enable people to use the old version if that is what they want.

v4hn, 2020-08-19:
moveit's `noetic-devel` branch is currently @rhaschke's personal playground. Nothing to resolve there really. :-)

tylerjw, 2020-08-19:
> moveit's `noetic-devel` branch is currently @rhaschke's personal playground. Nothing to resolve there really. :-)

I realize that now, it just confused me.

v4hn, 2020-08-19:
> Noetic is the last ROS1 release so we can no longer rely on future ROS1 releases to provide opportunity for breaking changes and will just have to deal with and accept that that will be a part of [...]

With this thought in might, it seems a very good idea to add SONAMES to all moveit-related repositories which do not have them already! That's probably not relevant for this repository though.

tylerjw, 2020-08-19:
What do you mean you mean by adding SONAMES?  I'm sorry, I don't follow.

v4hn, 2020-08-19:
https://github.com/ros-planning/moveit/pull/383 and https://github.com/ros-planning/geometric_shapes/issues/146 for reference.

tylerjw, 2020-08-19:
Reading about SONAME and that it is the standard way to do backwards-incompatible releases of library binaries in linux... How would we go about that for a messages repo?

v4hn, 2020-08-19:
On Wed, Aug 19, 2020 at 08:01:47AM -0700, Tyler Weaver wrote:
> Reading about SONAME and that it is the standard way to do backwards-incompatible releases of library binaries in linux... How would we go about that for a messages repo?

As I said, it's not relevant for *this* particular repository. But for pretty much all other moveit-dependencies you plan to release soon. :-)

tylerjw, 2020-08-19:
Do you know why `0.7.3` was the binary version that was used in moveit?

v4hn, 2020-08-19:
Please look at the links above again.

I only hardcoded 0.7.3 (as the MoveIt version in that branch back then) in `indigo-devel`.
`master` currently changes sonames *with each release*.
While this is not ideal, it means we don't have to bother with detailed ABI checks during releases.

tylerjw, 2020-08-19:
Ok, I'll go around and create PRs for any repo that doesn't have this (also need to validate that all the projects in moveit itself have these).

tylerjw, 2020-08-19:
Here is the SONAME change for geometric_shapes: https://github.com/ros-planning/geometric_shapes/pull/157

tylerjw, 2020-10-09:
This is released for Noetic, closing.

closed 2020-10-09 by tylerjw