moveit/moveit_msgs #138: PositionIKRequest has not member named attempts

robberthofmanfm, 2022-03-15:
Hello there,

I am trying to get a workspace that was written for melodic running on noetic. I've encountered the following error:

 `error: ‘moveit_msgs::PositionIKRequest’ {aka ‘struct moveit_msgs::PositionIKRequest_<std::allocator<void> >’} has no member named ‘attempts’`

Apparently, https://github.com/ros-planning/moveit_msgs/commit/82cda0ee22a06965bceb93ab0228a0cedbbd29eb removed that field.
Why?

robberthofmanfm, 2022-03-15:
Ah, seems like I have missed PR https://github.com/ros-planning/moveit_msgs/pull/76

JafarAbdi, 2022-03-16:
Hi @robberthofmanfm,

See https://github.com/ros-planning/moveit/pull/1288 for why it was changed & https://github.com/ros-planning/moveit/blob/master/MIGRATION.md#ros-melodic for how to migrate your kinematics.yaml

closed 2022-03-16 by JafarAbdi