moveit/moveit_msgs #142: Same ROS distribution in Windows and Ubuntu (Noetic) but different md5

Sission, 2022-07-20:
Hi! I've been using ROS on Ubuntu for years and it works great. Recently I have to use ROS in Windows so I followed this tutorial http://wiki.ros.org/Installation/Windows. It works fine and they can communicate between two computers (export ROS_MASTER_URI..)

However, when I tried to use Moveit, it gives me the following error:

` Client [/move_group] wants topic /move_group/goal to have datatype/md5sum [moveit_msgs/MoveGroupActionGoal/152e336e337dce7cbe639f1bd9c65def], but our version has [moveit_msgs/MoveGroupActionGoal/6514728724a6f4ce3f2f5a26db255bc9]. Dropping connection.`

Clearly something not compatible here. Then I find that, in windows:

```
F:\ws>rosmsg md5 moveit_msgs/MoveGroupActionGoal
6514728724a6f4ce3f2f5a26db255bc9
```
However, in Ubuntu:

```
rosmsg md5 moveit_msgs/MoveGroupActionGoal
152e336e337dce7cbe639f1bd9c65def
```

It seems like Ubuntu has the correct md5sum since it matches my other Ubuntu PC with ROS noetic.
I am not sure what happened. They are both Noetic versions but how can the msg md5 not be the same...

simonschmeisser, 2022-07-20:
Wild guess: line endings are different and maybe ros on windows uses windows line endings

mikeferguson, 2022-07-20:
> Wild guess: line endings are different and maybe ros on windows uses windows line endings

The MD5 is supposed to be computed without whitespace (I presume line endings are considered whitespace): http://wiki.ros.org/ROS/Technical%20Overview#Message_serialization_and_msg_MD5_sums

mikeferguson, 2022-07-20:
Are you sure you have the exact same version of the package on both systems? I'm fairly certain that moveit_msgs has had MD5-breaking changes within a distribution lifecycle - so you want to basically "roscd moveit_msgs" and see what the version number is in the package.xml and make sure they are the same (I'm not sure how the windows builds actually pick up new versions and thus how well sync'd they are with the Ubuntu Debian builds)

Sission, 2022-07-20:
> 

Hi! Thank you for your help!
Yes, I am 100% certain that they are both in the same version, which is Noetic. The Windows in built on a new PC and I've using Noetic on that Ubuntu machine for a while. Just to confirm, I type the following in Windows
```
F:\ws>roscd moveit_msgs

c:\opt\ros\noetic\x64\share\moveit_msgs>
```

I have tried all solutions I found online, including:

- delete the devel and build folder and rebuild
- manually change the md5sum in the MoveGroupActionGoal files (I know it is not the correct way to do it)
- delete Noetic totally and reinstall
- delete Noetic and try Melodic (give me another totally different md5)

To be honest, I was using that Ubuntu machine (PC1) to communicate with another real-time kernel Ubuntu machine (PC2), which controls the real robot. Since I have to get the data from the new Windows machine (PC3), I was wondering to communicate between this Windows PC (PC3) and the real-time kernel PC (PC2) directly and not need to use PC1. The main concern is that I want to keep the lag as low as possible. After reading http://wiki.ros.org/ROS/Tutorials/MultipleMachines, I guess I can bypass this problem by sending data from the Windows machine (PC3) to PC1 then PC1 can communicate with PC2, since they said **"A well-written node makes no assumptions about where in the network it run"**?

Anyways, this problem remains for me. A solution will be very appreciated.

mikeferguson, 2022-07-20:
Noetic is not a "version" - it's a "distribution". Then the packages released have a version in package.xml which is incremented every time a new version is released - this version number will be something like 0.11.2 (the latest version of moveit_msgs in the master branch).

Sission, 2022-07-20:
> Noetic is not a "version" - it's a "distribution". Then the packages released have a version in package.xml which is incremented every time a new version is released - this version number will be something like 0.11.2 (the latest version of moveit_msgs in the master branch).

Thank you for your clarification!
Upon further checking, it is indeed in different version.
In windows, it is version 0.11.1
However, in Ubuntu, it is version 0.11.2

Can I ask how to update moveit_msgs in Windows from 0.11.1 to 0.11.2?

mikeferguson, 2022-07-20:
I'm not really sure how Windows release/build works - so I don't know when the Windows build will be updated.

As a temporary workaround, you could check out moveit_msgs and build it on your Windows machine (as long as you are only using the messages - if you are also using MoveIt or the RVIZ plugins on the windows machine, you'd probably have to build the correct versions of those as well).

Sission, 2022-07-20:
> I'm not really sure how Windows release/build works - so I don't know when the Windows build will be updated.
>
> As a temporary workaround, you could check out moveit_msgs and build it on your Windows [...]

Thanks! I've tried to replace the corresponding folders in the moveit_msgs with the latest version but no luck. The moveit_msgs in Windows is slightly different than this repository. There are a "cmake" and "hook" folder and some other package files with different extensions. I do not think the message itself changed but the way it generates md5sum may be different between different versions or different OS. It may in a different folder. I also git clone this repository to my workspace and then build it. Remove the moveit_msgs inside the opt folder then confirm it runs from my workspace by "roscd moveit_msgs". Sadly it still gives me the previously same md5sum.

mikeferguson, 2022-07-20:
It looks like the message did change between 0.11.1 and 0.11.2 - a pipeline_id was added to the motion planning request - https://github.com/ros-planning/moveit_msgs/commit/f42f12433ea8f029ef6d2653b8b42e3cf8f7a066

I'd suggest the best course of action is to reach out to whoever maintains the Windows builds and ask them how to get the newer version

rhaschke, 2022-09-09:
Closing as the culprit was a version mismatch.

closed 2022-09-09 by rhaschke