moveit/moveit_msgs #156: Divergence of MotionPlanRequest.msg between ROS 1 and 2 Versions

acbuynak, 2023-04-01:
Stumbled across a breaking change between versions of `msg/MotionPlanRequest.msg` while building a ROS Bridge. Reading through Issues/PR's it looks like there's an intentional history. Adding this issue to highlight the difference.

### Brief History:
#113, message field was added, but wrong name used.
#122, attempts to correct field names.
#130, rewrite of 122, successfully merged into `master`.

All ROS 2 versions appear to be 9 commits behind, critically f562cc8b499d19c447babb6eec271fd7dbf0ece3

I'm unfamiliar with the ROS 2 moveit ecosystem to say the appropriateness of whether it's okay for these to match. At this time, users attempting to bridge MoveIt messages are unable to do so without modifying one side or the other. Given ROS Noetic is around until 2025, it may be useful to have them match.

methylDragon, 2024-08-02:
I seem them here: https://github.com/moveit/moveit_msgs/blob/9756797dcbc293c7cae66541a7bc0c7acec85892/srv/GetCartesianPath.srv#L48-L49

is this still an issue?

acbuynak, 2024-08-12:
This issue was pedantic documentation of changes to the `MotionPlanRequest.msg` type from ROS 1 -> 2 versions, and subsequent renaming between ROS 2 branches. Specifically, with commits 3f98f5d492474331099238e2d2521a3cc5774893 and subsequent renaming by 4a1e8568836bf668dad8062889d293d5429bb5db on the latest `ros2` branches.

(ros2)
https://github.com/moveit/moveit_msgs/blob/c713195687cc4afcd466062007cc97cb1736638e/msg/MotionPlanRequest.msg#L57-L58

(humble)
https://github.com/moveit/moveit_msgs/blob/99e6898f589d3d6929c6da68e0c3f1057866aca8/msg/MotionPlanRequest.msg#L57-L58

(melodic-devel)
```...Message did not include this variable.```

Bit nuanced, but I documented it here. 🤷  This issue can be closed as no change.

closed 2024-08-12 by acbuynak