moveit/moveit_msgs #185: Cartesian control for a new servo implementation.

rafal-gorecki, 2025-02-28:
Hello,
I have a 4 DOF robot. And I would like to control using a servo using Cartesian coordinates. When I use the `servo_node/delta_twist_cmds` topic, I am not able to perform front-back movements that should be possible without any problems with end_of_efector. Instead, he receives a warning 
`Very close to a singularity, emergency stop`

In the Humble version I was able to use the ChangeDiffDimensions messages to make possible moves, but now I don't know if the package now assumes that all freedom steps are possible when receiving a msg. Related with: https://github.com/moveit/moveit2/pull/2224

Can you give me some advice?

AndyZe, 2025-02-28:
Maybe we should bring that feature back. It was very convenient. Of course it will only work with the Jacobian IK solver. Are you up for making a PR to bring it back?

AndyZe, 2025-03-01:
You could also try switching to the bio_ik kinematics solver as described here:  https://moveit.picknik.ai/main/doc/examples/realtime_servo/realtime_servo_tutorial.html#inverse-kinematics-in-servo

bio_ik allows approximate solutions, which is necessary for a 4-dof arm.

This should be easy to try. I'm not sure how good the performance would be so let us know if you do try it.

rafal-gorecki, 2025-03-11:
I wonder if it makes sense to restore this functionality. It was changed for some reason, so it's hard for me to say it is necessary to revert. If it worked when using IK, it would also be satisfactory for me. However, I can't achieve this either after changing to `bioIK` or with `kdl` with the `position_only_ik: true`

```bash
[moveit.moveit.kinematics.kdl_kinematics_plugin]: Using position only ik
[moveit.moveit.ros.servo]: Very close to a singularity, emergency stop
```

AndyZe, 2025-03-11:
We removed it in 2023 as part of this project:  [https://picknik.ai/moveit/benchmarking/inverse%20kinematics/servo/2023/11/21/GSoC-2023-MoveIt-Servo-and-IK-Benchmarking.html](https://picknik.ai/moveit/benchmarking/inverse%20kinematics/servo/2023/11/21/GSoC-2023-MoveIt-Servo-and-IK-Benchmarking.html)

We thought that bio_ik or one of the other IK solvers that allows approximate IK solutions / weighted cost functions would be a good replacement, but it sounds like that wasn't true.

I bet you could get it working if you used the right combination of `bio_ik` goals. I don't know how to specify those off the top of my head, however.