moveit/moveit_msgs #187: Missing release

urfeex, 2025-07-04:
Hi, https://build.ros2.org/job/Kbin_uN64__moveit_ros_move_group__ubuntu_noble_amd64__binary/19/console indicates that a release containing #186 should be made :-)

mikeferguson, 2025-07-14:
Status pages are the best way to track things - you can search by "moveit" as well: https://repo.ros2.org/status_page/ros_rolling_default.html

(I also struggle to find these, so I have the full list of relevant status pages linked at the bottom of my cookbook: https://github.com/mikeferguson/ros2_cookbook?tab=readme-ov-file)

gavanderhoorn, 2025-07-14:
They're linked from their respective `docs.ros.org` pages:

[image]

[This one](https://docs.ros.org/en/jazzy/Installation/Ubuntu-Install-Debs.html#resources) for Jazzy on Ubuntu Noble fi.

nbbrooks, 2025-07-14:
I am trying to do the Subsequent Release instructions without a GH classic token with `workflow` scope (since the ROS docs were written, I believe workflow scope now requires full private repo access which is not acceptable). I though I had done the previous moveit release with only workflow scope, but not sure how I got around this set of errors in `bloom-release`

```
 ! [remote rejected] debian/kilted/bookworm/moveit_msgs -> debian/kilted/bookworm/moveit_msgs (refusing to allow a Personal Access Token to create or update workflow `.github/workflows/format.yml` without `workflow` scope)
```

The [ros2-gbp/moveit_msgs-release/](https://github.com/ros2-gbp/moveit_msgs-release/branches) branches are pushed up but I do not know what steps to do next manually,

Has anyone done this last part manually that the workflow normally does?